emsec/hal #613: Display predecesor of a net or pin.

Blebowski, 2025-05-12:
**Describe the solution you'd like**
Ability to trace (add to displayed schematic) a driver of a net or a single pin.

**Describe alternatives you've considered**
Currently, there is a "display predecesor" with similar functionality.
This works great, but displays predecessors for all input pins of a gate.
It would be handy if the predecessor (driver) was added to the schematic
view upon e.g. double clicking on the net.

When tracing a path (e.g. a timing path), it may be handy to display a single
predecessor so that in the path ends as "a chain" in the window.

joern274, 2025-05-13:
What about
1. Isolate gate in new view
2. Navigate with Arrow Left + Arrow Up/down to desired input pin
3. Arrow Left to add net source (driver) to isolated view. Popup should appear if multi-driven net.
?

joern274, 2025-05-21:
@Blebowski , did it work?

Blebowski, 2025-05-21:
I did not know it works like that. Tried it and the navigation is sufficient. The UI is slightly different than GateVisionPRO, but intuitively usable. Thanks !

closed 2025-05-21 by Blebowski