emsec/hal #614: Unable to complete Make process on Ubuntu 22.04 due to Resynthesis Plugin

UsuallyReadsTheManual, 2025-05-15:
OS - Ubuntu 22.04

Downloaded the tar source file for hal 4.4.1. 

Expected result of make -$(nproc) should yield completed build.

Actual results: make: *** [Makefile:156: all] Error 2
Due to: 

Command ran for cmake:  cmake .. -DPL_RESYNTHESIS=OFF
Have also tried -CMAKE_DPL_RESYNTHESIS=OFF and variants of =0
cmake seems to be ignoring the directive to ignore resynthesis plugin during make/cmake process.

Actual error from build process:

```
/hal-4.4.1/plugins/netlist_preprocessing/src/netlist_preprocessing.cpp:16:10: fatal error: resynthesis/resynthesis.h: No such file or directory
   16 | #include "resynthesis/resynthesis.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
``` 

Here is the output of the cmake process:
```
-- Selected build type: Release
-- ccache found!
-- System is:
-- Linux distro is: Ubuntu
-- Linux Distro version name is: jammy
-- System Arch is: x86_64
-- Extracting Version Information in PATH: /opt/hal-4.4.1
-- HAL_VERSION_RETURN:
-- HAL_VERSION: v4.4.1
-- Checking for module 'bitwuzla'
--   Package 'bitwuzla', required by 'virtual:world', not found
-- Bitwuzla not found, but this is optional...
-- openMP found
-- RapidJSON found. Headers: /usr/include
-- Found rapidjson /usr/include
-- Set rapidjson path successully: /usr/include
-- Found pybind11: /usr/include (found version "2.9.1")
-- using spdlog from deps
-- using subprocess from deps
-- using nlohmann_json from deps
-- Python3_INCLUDE_DIRS: /usr/include/python3.10
-- Python3_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.10.so
-- PYTHON_MODULE_PREFIX:
-- PYTHON_MODULE_EXTENSION: .cpython-310-x86_64-linux-gnu.so
-- Checking for module 'libcgraph'
--   Package 'libcgraph', required by 'virtual:world', not found
-- Checking for module 'libcdt'
--   Package 'libcdt', required by 'virtual:world', not found
-- Checking for module 'libgvc'
--   Package 'libgvc', required by 'virtual:world', not found
-- Checking for module 'libpathplan'
--   Package 'libpathplan', required by 'virtual:world', not found
-- Could NOT find Graphviz (missing: GRAPHVIZ_LIBRARIES GRAPHVIZ_INCLUDE_DIR)
-- ABC not found
-- Will build abc ourselves, check README.md to see how to speed up the process...
-- Found z3
--     Z3_LIBRARIES: /usr/lib/x86_64-linux-gnu/libz3.so
--     Z3_INCLUDE_DIRS: /usr/include
-- Using igraph from /opt/hal-4.4.1/deps/igraph-0.10.x
-- Checking supported compiler flags...
-- Performing Test supports_-Wall
-- Performing Test supports_-Wall - Success
-- Performing Test supports_-Wextra
-- Performing Test supports_-Wextra - Success
-- Performing Test supports_-pedantic
-- Performing Test supports_-pedantic - Success
-- Performing Test supports_-Wshadow
-- Performing Test supports_-Wshadow - Success
-- Performing Test supports_-g
-- Performing Test supports_-g - Success
-- Performing Test supports_-Wno-undef
-- Performing Test supports_-Wno-undef - Success
-- Performing Test supports_-Werror=return-type
-- Performing Test supports_-Werror=return-type - Success
-- Performing Test supports_-fcolor-diagnostics
-- Performing Test supports_-fcolor-diagnostics - Failed
-- Performing Test supports_-fdiagnostics-color=always
-- Performing Test supports_-fdiagnostics-color=always - Success
-- Performing Test supports_-fsized-deallocation
-- Performing Test supports_-fsized-deallocation - Success
-- Performing Test supports_-O0
-- Performing Test supports_-O0 - Success
-- Performing Test supports_-g
-- Performing Test supports_-g - Success
-- Performing Test supports_-O3
-- Performing Test supports_-O3 - Success
-- Performing Test supports_-DNDEBUG
-- Performing Test supports_-DNDEBUG - Success
-- Performing Test supports_-Os
-- Performing Test supports_-Os - Success
-- Performing Test supports_-DNDEBUG
-- Performing Test supports_-DNDEBUG - Success
-- Performing Test supports_-O2
-- Performing Test supports_-O2 - Success
-- Performing Test supports_-g
-- Performing Test supports_-g - Success
-- Performing Test supports_-O1
-- Performing Test supports_-O1 - Success
-- Performing Test supports_-g
-- Performing Test supports_-g - Success
-- Performing Test supports_-fno-inline-functions
-- Performing Test supports_-fno-inline-functions - Success
-- Performing Test supports_-fno-inline-functions-called-once
-- Performing Test supports_-fno-inline-functions-called-once - Success
-- Performing Test supports_-fno-optimize-sibling-calls
-- Performing Test supports_-fno-optimize-sibling-calls - Success
-- Performing Test supports_-fno-omit-frame-pointer
-- Performing Test supports_-fno-omit-frame-pointer - Success
-- HAL_STUDY mode turned OFF!
-- GATE_LIBS: /opt/hal-4.4.1/plugins/gate_libraries/definitions/.gitignore;/opt/hal-4.4.1/plugins/gate_libraries/definitions/NanGate_15nm_OCL.hgl;/opt/hal-4.4.1/plugins/gate_libraries/defi>
-- HELPER_GATE_LIBS: /opt/hal-4.4.1/plugins/gate_libraries/definitions/helper_libs/aoixm_hal_i4.hgl;/opt/hal-4.4.1/plugins/gate_libraries/definitions/helper_libs/im_hal_i4.hgl
-- Using Qt version 5
-- Using quazip from /opt/hal-4.4.1/deps/quazip-1.3
-- CPACK_GENERATOR:
-- PROJECT_ARCH: x86_64
-- Building for focal
-- version-abc: 4.4.1
-- version: 4.4.1
-- Debian version: 4.4.1-ppa1~focal1
-- CPACK_SOURCE_IGNORE_FILES: /mxe/;/\\.vscode/;/\\.codelite/;/\\.idea/;/cmake-.*/;\\.project$;\\.workspace$;\\.iml$;\\.mk$;\\.astylerc$;\\.editorconfig$;/Makefile$
-- Upload PPA is OFF
-- SIMULATE_DPUT_STR is -s
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/hal-4.4.1/build
```

is cmake blatantly ignoring option inputs??
Any assistance is greatly appreciated, thank you!

UsuallyReadsTheManual, 2025-05-15:
Edit: tried with cmake ..-DBUILD_ALL_PLUGINS=ON
This worked.

Issue seems akin to [issue 588](https://github.com/emsec/hal/issues/588 ) ?

UsuallyReadsTheManual, 2025-05-15:
Unfortunately some additional issues now that the initial build process worked. I'd like to build the .deb package if possible but I cannot:

```
dpkg-checkbuilddeps: error: Unmet build dependencies: libigraph0-dev (>= 0.7.1) qt5-default (>= 5.7)
```
```
make[2]: *** [packaging/CMakeFiles/debuild_focal.dir/build.make:78: Debian/focal/hal-reverse_4.4.1-ppa1~focal1_source.changes] Error 29
make[1]: *** [CMakeFiles/Makefile2:2334: packaging/CMakeFiles/debuild_focal.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

qt5-default was dropped prior to ubuntu 22.04 .  base, and dev tools for qt5 are installed but this doesn't seem to matter to the build process. Also for libigraph, shouldn't that get built as part of the make process in the first place? The source gets pulled in? The version in the ubuntu repo is lower than the version expected, and even making and then doing a make install on igraph doesn't seem to satisfy it either?

joern274, 2025-05-16:
Sorry, must admit that the .deb package has not been maintained for quite some time. It is on the TODO list. 

I am glad you found a workaround for the Resysnthesis problem originally reported. I will look into the plugin dependency list. Obviously some dependencies are not reflected by the build process.

UsuallyReadsTheManual, 2025-05-16:
Also I should note, even with having manually downloaded the expected tar for the igraph source and placing it in the deps folder, I had to chain together a cmale, followed by make, copying over the cmakelist and extracted contents from the tar to the build/igraph/download and hal/deps/igraph.../ Folder, before that would complete a build properly.

I also had to feed a url and hash to one of the files (not in front of me sorry) that called for downloading the package if not present as the build output mentioned finding it but not having a matching hash.

For offline building it would be great to have a way to allow direct turn off/on of downloading and such. I tried using the use vendor package option but it had no effect.

joern274, 2025-05-17:
The reported Resysnthesis <-> Netlist preprocessing dependecy problem in version 4.4.1 (tarball) has been fixed (actually by me, I just forgot) on October 15th last year. We should issue a new release soon so that people who prefer installing from tarball rather than by `git clone` can also benefit from that fix.

I am not sure I understand the IGRAPH problems you experienced. Once I unpacked the tarball on an Out-of-the-box-Ubuntu system the IGRAPH library was downloaded, built and installed automatically during the HAL `make`. You cannot use the IGRAPH package from Ubuntu since Ubuntu ships a rather old version and there was a major API break in IGRAPH. Or did I misunderstand and you were downloading a specific IGRAPH version manually on purpose? That could of course interfere with the build script.

The third topic - not building the .deb package - was already mentioned in other issues and on our TODO list. I would therefore like to close this issue unless I get more detailed information how to reproduce the IGRAPH problem.

joern274, 2025-05-22:
You are right, the issue #592 (`make install` not working correctly) was never fixed. Most people were starting HAL from the build directory skipping the installation, thus other issues were more urgent. But I am working on the install issue right now.

My preliminary findings are that there is a cmake statement missing setting the RPATH/RUNPATH when installing. You can verify it when you run  `ldd /usr/local/lib/hal_plugins/module_identification.so`  or `readelf -d /usr/local/lib/hal_plugins/module_identification.so`.

Not sure why ldconfig doesn't resolve the issue. I was setting `LD_LIBRARY_PATH` as workaround and with that all dependencies were resolved. Does that work in your place ?