emsec/hal #615: Cell symbol

Blebowski, 2025-05-21:
Hi,

it would be really great if there was a way on how to display cells as symbols, not simply
as rectangles. That-way the visualized circuit would functionally resemble what it actually does.

There may be two ways on how to achieve this:
1. Load some type of symbol file explicitly.

There are couple of symbol formats (e.g. XSCHEM has one, then there is proprietary OA format,
and other plaintext formats exist). This might easy way on getting this done, but IMHO not the best.

2. Infer the symbol from the known logic function by HAL itself.

There could be set of predefined pictures for AND, OR, XOR, FF, LATCH, MUX, and HAL would 
form the symbol based on parsing the known logic function. If the logic function is not known
(e.g. as in case of black-box macro as I was mentioning in the other issue), HAL would simply
display rectangle as now.

joern274, 2025-05-22:
It is an interesting idea, I like the spirit of innovation. It would need some more input and sample drawings before we can consider implementing it. 

- How would the pins get aligned in a non rectangular shape?
- How can the shape gets automatically extended to make space for lengthy names and labels?
- Right now all input- and output nets connecting a gate start at the same x position. Yet wiring is a complicated task requiring a lot of CPU time. We would have to test whether it causes a huge setback in layout performance if all wires start/end at different positions.
- Will non rectangular shape have an impact on performance of zoom- or pan-operations?

And then there are some NO-GO's to consider. I have seen some FF graph representations where Set is connected from above (top side) and Reset from below (bottom side). That would screw up everything in our layout algorithm. Inputs must be on left side, outputs to the right.

Blebowski, 2025-05-25:
I made some hand-drawings:

[image]

The idea is that everything would stick to some grid. The picture above shows N with grid dimension and each pin is placed either at N or at N /2. If halving the grid, everything would stick to N -> All pins at known positions. The bounding box (the dashed red), is always rectangle, and it is a place where the wires would stick.

If there are more complex combo cells, their "picture" can be assembled from the primitive "AND", "OR", "XOR" and the "inverting circle" before / after a pin. Such cells can be assembled in "stages" as they propagate through the primitive cells (e.g. AOI cells).
I am sure there will be some corner-cases, where the sub-cell placement within complex cell may be problem, but IMHO this would cover 95 percent of cells available. 

Some topics that may be TODO in this concept:
- How to handle multi-bit flops ?
- An "ADD" and "SUB" drawing with + / - in the middle may be good to add, I have seen those in some PDKs.

RenWal, 2025-09-26:
Thanks Jörn! I'll put this on the agenda for our next meeting, let's have a quick chat on how we can build this together :)

Blebowski, 2026-01-02:
Hi @joern274 ,

thanks, I finally got to trying the implementation.

The simple example works, I can see basic combinatorial gate icons.

[image]

Is it supposed to have the surrouing gate icon line so thick ? Or am I abusing something in the QT configuration ?
How can I bring this up on a custom netlist ?